BatchDrake/suscan #11: Undefined reference to `su_channel_detector_sample_was_consumed'

sh123, 2018-04-02:
Tried to build from master, got two linker errors

```
analyzer/inspector.c:373: undefined reference to `su_channel_detector_sample_was_consumed'
analyzer/inspector.c:380: undefined reference to `su_channel_detector_get_last_sample'
```

BatchDrake, 2018-04-02:
My fault, `master` is outdated w.r.t sigutils' master. Try to build from
`feature/fsk-inspector` instead.

If you still have trouble compiling it and don't mind waiting, I'm planning
a big merge this evening (UTC+2). I'll keep you updated.

BatchDrake, 2018-04-02:
> Gonzalo José Carracedo Carballal

Hi again,

I've just updated both sigutils and suscan. Just clone from master, run
./autogen.sh in both projects and compile as usual. It should work now.

2018-04-02 9:25 GMT+02:00 Gonzalo José Carracedo Carballal <
batchdrake@gmail.com>:

closed 2023-04-12 by BatchDrake