BatchDrake/suscan #12: suscan: main/main.c:175: suscan_gui_disconnect: Assertion `gui->analyzer != NULL' failed.

chris4git, 2018-09-02:
Hi,
very interesting tool ! 
unfortunately it crashes before doing anything useful :-(
Last traces :
------
(suscan:7078): Gdk-CRITICAL **: 09:23:58.208: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:7078): Gdk-CRITICAL **: 09:23:58.208: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:7078): Gdk-CRITICAL **: 09:23:58.208: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:7078): Gdk-CRITICAL **: 09:23:58.208: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:7078): Gdk-CRITICAL **: 09:23:58.208: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:7078): Gdk-CRITICAL **: 09:23:58.209: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:7078): Gdk-CRITICAL **: 09:23:58.209: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed
suscan: main/main.c:175: suscan_gui_disconnect: Assertion `gui->analyzer != NULL' failed.
Abort (core dumped)
--------

the crash is after displaying a spectrum.
OS : Ubuntu Mate 18.04.1 LTS 
libgtk-3 3.22.30-1ubuntu1

Thanks for your help

BatchDrake, 2018-09-02:
Hi, and thank you very much for your feedback! Can you provide details on
the signal source you were using? Can you provide a detailed list of steps
to reproduce this crash?

Thanks!

chris4git, 2018-09-02:
pulled, rebuilt and no more crash :-)
but that kind of errors :

(suscan:23349): Gdk-CRITICAL **: 23:05:35.096: gdk_window_create_similar_surface: assertion 'GDK_IS_WINDOW (window)' failed

(suscan:23349): Gtk-CRITICAL **: 23:05:42.405: gtk_widget_grab_default: assertion 'gtk_widget_get_can_default (widget)' failed
Looking up gain TUNER in 7 gains
Found Elonics E4000 tuner
[INFO] Using format CF32.
Looking up gain TUNER in 7 gains
Found Elonics E4000 tuner
[INFO] Using format CF32.

(suscan:23349): Gtk-CRITICAL **: 23:11:53.886: gtk_widget_grab_default: assertion 'gtk_widget_get_can_default (widget)' failed

BatchDrake, 2018-09-03:
These errors are well known, and they are probably related to the high Xorg CPU usage during capture. I'm closing this issue, but I'm opening a new one to fix these warnings once for all.

closed 2018-09-03 by BatchDrake