BatchDrake/suscan #15: Is a waterfall expected ?

chris4git, 2018-09-04:
Hi,
I try to run SUScan with a rtl-sdr stick as default source (@300000 sps).
When I click on the start button a spectrum line is run, a lot of CPU is used by Xorg but nothing moves anymore on the screen except, sometimes, some red lines/rectangles (see attached screenshot).
No waterfall and the window seems stuck :-(
The netbook used is not very powerful (AMD C-70) but it runs gqrx with a spectrum display, a waterfall and its demodulators just fine :-/

Any hints ?

BatchDrake, 2018-09-04:
This is a well known GTK 3/Cairo issue that I'm struggling to solve, and I
got no luck so far. I have two options here: I either do all the scaling
and drawing by myself directly in C, or I give up on GTK 3 and use QT
instead. And the first option does not look attractive at all.

For now, I would not use Suscan above the 2Msps (lame, I know). My next
task will be to fix this annoying problem once for all.

chris4git, 2018-09-04:
thanks for your quick reply.
In this case I use it at only 0.3Msps !
...gqrx use qt5...

BatchDrake, 2019-08-16:
I finally did what you say. I ended up giving up on the GTK interface and started from scratch with Qt5. 6 Msps and realtime demodulation with audio playback. What do you think?

BatchDrake, 2023-04-12:
I am closing this issue as Suscan does not have an UI anymore. Try SigDigger instead, it beats what Suscan used to be.

closed 2023-04-12 by BatchDrake