BatchDrake/suscan #16: Can't test suscan library

andruxa-smirnov, 2019-08-16:
I build dependencies (suscan) and install it. When i run "suscan", to test installation, i received next message: "suscan: error while loading shared libraries: libsigutils.so.0: cannot open shared object file: No such file or directory". Previously i build sigutils and install it too.
Ubuntu 18.04, x64.

BatchDrake, 2019-08-16:
This seems a PATH problem. Can you try the following:

```
% export LD_LIBRARY_PATH="/usr/local/lib:$LD_LIBRARY_PATH"
% suscan
```

miek, 2019-08-17:
You may just need to run `sudo ldconfig` to update the linker cache.

andruxa-smirnov, 2019-08-17:
> This seems a PATH problem. Can you try the following:
> 
> ```
> % export LD_LIBRARY_PATH="/usr/local/lib:$LD_LIBRARY_PATH"
> % suscan
> ```

Yes. That is proper solution. Thanks!

closed 2019-08-17 by andruxa-smirnov