BatchDrake/suscan #18: SigDigger appimage audio clipping at 32,000 up

2WR3505, 2019-08-21:
Wavin drake, i was testing this on my skywave linux install, and the audio gets choppy at 32,000 and up. is there any thing i can do to fix this by tweakin settings in the program?

2WR3505, 2019-08-21:
ah yea, that helped alot, takin the audio volume to 6% cleaned it up alot. i can actually hear and understand myself and the other guys talking now, lol.

i didnt even need to mess with the frequency bandwidth :)

closed 2019-08-22 by BatchDrake

BatchDrake, 2019-08-22:
@2WR3505 I think I fixed the main issue here, but now I'm dealing with a new one related to this. I also detail a workaround in the mean time: https://github.com/BatchDrake/suscan/issues/20

I'm building an AppImage with some bugfixes, including this.

BatchDrake, 2019-08-22:
You can download the new AppImage now in https://github.com/BatchDrake/SigDigger/releases/tag/0.0.6-alpha

commit ab4f9537 (2019-08-22): Prevent clipping in AM demodulator (partially fixes #18)