BatchDrake/suscan #19: Crash into analyze file

andruxa-smirnov, 2019-08-21:
When i start Open Inspector for IQ-rec file, app crash with messages:
"Gtk-Message: 22:38:03.538: GtkDialog mapped without a transient parent. This is discouraged.
"Spectrum zoom: 1.1x"
"Spectrum zoom: 1.2x"
"Spectrum zoom: 1.4x"
"Spectrum zoom: 1.5x"
"Spectrum zoom: 1.7x"
"Spectrum zoom: 1.9x"
"Spectrum zoom: 2.1x"
"Spectrum zoom: 2.2x"
"Spectrum zoom: 2.5x"
"Spectrum zoom: 2.7x"
"Spectrum zoom: 3.0x"
"Spectrum zoom: 3.3x"
"Spectrum zoom: 3.7x"
"Spectrum zoom: 4.1x"
"Spectrum zoom: 4.4x"
"Spectrum zoom: 4.9x"
"Spectrum zoom: 5.5x"
"Spectrum zoom: 5.6x"
"Spectrum zoom: 5.1x"
"Spectrum zoom: 4.6x"
"Spectrum zoom: 3.6x"
double free or corruption (fasttop)double free or corruption (fasttop)double free or corruption (fasttop)"

Thats happen not always (memory leaking, may be)

BatchDrake, 2019-08-22:
@andruxa-smirnov do you mind pulling both Suscan and SigDigger and rebuilding again, just to check that everything works now?

closed 2019-08-22 by BatchDrake

andruxa-smirnov, 2019-08-22:
> @andruxa-smirnov do you mind pulling both Suscan and SigDigger and rebuilding again, just to check that everything works now?

Yes, no mistake in updated code

commit 2af94a11 (2019-08-22): Improper ownership handling of message objects in worker thread (probably fixes #19)