BatchDrake/suscan #30: Problems with building on Arch Linux

dvdesolve, 2019-12-30:
Problems described here are pretty much the same as in BatchDrake/sigutils#14 so I will post here only the relevant messages.

1. Linking problem
```
==> Making package: suscan-git r479.9250ea0-1 (Tue 31 Dec 2019 02:41:10 AM MSK)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning suscan-git git repo...
Cloning into bare repository '/home/viktor/build/aur/suscan-git/suscan-git'...
remote: Enumerating objects: 269, done.
remote: Counting objects: 100% (269/269), done.
remote: Compressing objects: 100% (174/174), done.
remote: Total 3733 (delta 169), reused 180 (delta 93), pack-reused 3464
Receiving objects: 100% (3733/3733), 2.52 MiB | 4.11 MiB/s, done.
Resolving deltas: 100% (2863/2863), done.
==> Validating source files with md5sums...
    suscan-git ... Skipped
==> Extracting sources...
  -> Creating working copy of suscan-git git repo...
Cloning into 'suscan-git'...
done.
==> Starting pkgver()...
==> Updated version: suscan-git r480.29625cb-1
==> Starting build()...
-- The C compiler identification is GNU 9.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'sigutils>=0.1'
--   Found sigutils, version 0.1
-- Checking for module 'sndfile>=1.0.2'
--   Found sndfile, version 1.0.28
-- Checking for module 'fftw3f>=3.0'
--   Found fftw3f, version 3.3.8
-- Checking for module 'SoapySDR>=0.5.0'
--   Found SoapySDR, version 0.7.1
-- Checking for module 'libxml-2.0>=2.9.0'
--   Found libxml-2.0, version 2.9.10
-- Checking for module 'volk>=1.0'
--   Found volk, version 2.0
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/makepkg/suscan-git/src/suscan-git/build
Scanning dependencies of target suscan
[  2%] Building C object CMakeFiles/suscan.dir/util/confdb.c.o
[  4%] Building C object CMakeFiles/suscan.dir/util/cfg.c.o
[  6%] Building C object CMakeFiles/suscan.dir/util/deserialize.c.o
[  8%] Building C object CMakeFiles/suscan.dir/util/object.c.o
[ 10%] Building C object CMakeFiles/suscan.dir/util/serialize.c.o
[ 13%] Building C object CMakeFiles/suscan.dir/util/util.c.o
[ 15%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/inspector.c.o
[ 17%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/interface.c.o
[ 19%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/params.c.o
[ 21%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/ask.c.o
[ 23%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/audio.c.o
[ 26%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/fsk.c.o
[... 116 lines cut ...]
/tmp/makepkg/suscan-git/src/suscan-git/codec/codecs/diff.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/analyzer.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/bufpool.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/client.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/estimator.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspsched.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/insp-server.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/mq.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/msg.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/slow.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/source.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrc.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/symbuf.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/estimators/fac.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/estimators/nonlinear.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/cyclo.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/fmcyclo.c
>/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/fmspect.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/pmspect.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/timediff.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/exp-2.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/exp-4.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/exp-8.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/psd.c
```

BatchDrake, 2019-12-31:
Same procedure, I'd like to have a copy of your `CMakeFiles/*/link.txt` files inside your build directory.

Thanks!

dvdesolve, 2019-12-31:
[suscan.dir/link.txt](https://github.com/BatchDrake/suscan/files/4012824/link.txt)
[suscan.status.dir/link.txt](https://github.com/BatchDrake/suscan/files/4012825/link.txt)

dvdesolve, 2019-12-31:
Diff to fix:

```
--- CMakeLists.txt.old	2019-12-31 17:59:14.471227845 +0300
+++ CMakeLists.txt	2019-12-31 18:01:12.973696170 +0300
@@ -228,10 +228,15 @@
   PRIVATE . ${UTILDIR} ${CODECLIB_DIR} ${SRCDIR})
 
 # Required dependencies
 set_target_properties(suscan.status PROPERTIES COMPILE_FLAGS "${SIGUTILS_SPC_CFLAGS}")
 
+link_directories(suscan.status ${PROJECT_BINARY_DIR})
+target_link_libraries(suscan.status sigutils)
+target_link_libraries(suscan.status suscan)
+target_link_libraries(suscan.status m)
+
 target_include_directories(suscan.status SYSTEM PUBLIC ${SNDFILE_INCLUDE_DIRS})
 target_link_libraries(suscan.status ${SNDFILE_LIBRARIES})
 
 target_include_directories(suscan.status SYSTEM PUBLIC ${FFTW3_INCLUDE_DIRS})
 target_link_libraries(suscan.status ${FFTW3_LIBRARIES})
@@ -241,14 +246,10 @@
 
 target_include_directories(suscan.status SYSTEM PUBLIC ${XML2_INCLUDE_DIRS})
 target_link_libraries(suscan.status ${XML2_LIBRARIES})
 
 target_link_libraries(suscan.status ${CMAKE_THREAD_LIBS_INIT})
-link_directories(suscan.status ${PROJECT_BINARY_DIR})
-target_link_libraries(suscan.status sigutils)
-target_link_libraries(suscan.status suscan)
-target_link_libraries(suscan.status m)
 
 # Optional dependencies
 if(VOLK_FOUND)
   set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -DHAVE_VOLK=1")
   target_include_directories(suscan.status SYSTEM PUBLIC ${VOLK_INCLUDE_DIRS})
```

dvdesolve, 2020-01-02:
Latest commit, `makepkg` still complains about buildroot traces:
```
[23:24 viktor@desolve-nettop ~/build/aur/suscan-git]$ strings suscan.status | grep makepkg
/tmp/makepkg/suscan-git/src/suscan-git/src/fingerprint.c
[23:24 viktor@desolve-nettop ~/build/aur/suscan-git]$ strings libsuscan.so | grep makepkg
/tmp/makepkg/suscan-git/src/suscan-git/util/cfg.c
/tmp/makepkg/suscan-git/src/suscan-git/util/confdb.c
/tmp/makepkg/suscan-git/src/suscan-git/util/deserialize.c
/tmp/makepkg/suscan-git/src/suscan-git/util/object.c
/tmp/makepkg/suscan-git/src/suscan-git/util/serialize.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/inspector.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/interface.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/params.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/impl/ask.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/impl/audio.c
?/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/impl/fsk.c
yE>/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspector/impl/psk.c
/tmp/makepkg/suscan-git/src/suscan-git/codec/codec.c
/tmp/makepkg/suscan-git/src/suscan-git/codec/codecs/diff.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/analyzer.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/bufpool.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/client.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/estimator.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/inspsched.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/insp-server.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/mq.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/msg.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/slow.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/source.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrc.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/symbuf.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/estimators/fac.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/estimators/nonlinear.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/cyclo.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/fmcyclo.c
>/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/fmspect.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/pmspect.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/timediff.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/exp-2.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/exp-4.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/exp-8.c
/tmp/makepkg/suscan-git/src/suscan-git/analyzer/spectsrcs/psd.c
```

BatchDrake, 2020-01-03:
Strange, it's like the substitution is not working (maybe because there are symlinks in the build path). Could you try replacing the following line in CMakeLists.txt

```
set(CMAKE_C_FLAGS
"${CMAKE_C_FLAGS} -D__FILENAME__='\"$(subst ${CMAKE_SOURCE_DIR}/,,$(abspath $<))\"'")
```
With:

```
set(CMAKE_C_FLAGS 
"${CMAKE_C_FLAGS} -D__FILENAME__='\"$(subst $(realpath ${CMAKE_SOURCE_DIR})/,,$(realpath $<))\"'")
```

And try again? If this does not work either, we can conditionally define __FILENAME__ to the base name of the file when building in release mode.

dvdesolve, 2020-01-03:
Still doesn't work =\

BatchDrake, 2020-01-06:
The following should remove any reference to build paths in the source code. Replace in CMakeLists the following line:

```
set(CMAKE_C_FLAGS
"${CMAKE_C_FLAGS} -D__FILENAME__='\"$(subst ${CMAKE_SOURCE_DIR}/,,$(abspath $<))\"'")
```

With:

```
set(CMAKE_C_FLAGS
"${CMAKE_C_FLAGS} -D__FILENAME__='\"none\"'")
```
If it still does not work, it would mean that there is something fishy going on and I would have to reverse engineer your compiled library (libsuscan.so) and figure out where these paths are being referenced at.

dvdesolve, 2020-01-07:
Now I have a different problem during linking. I believe that this is related to the latest commits:
```
[100%] Linking C executable suscan.status
/usr/bin/ld: libsuscan.so: undefined reference to `su_channel_detector_get_iters'
/usr/bin/ld: libsuscan.so: undefined reference to `su_channel_detector_rewind'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/suscan.status.dir/build.make:130: suscan.status] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/suscan.status.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

BatchDrake, 2020-01-07:
Indeed, I've been working on it yesterday. Pull the latest `sigutils` and build again.

dvdesolve, 2020-01-07:
Tried with latest sigutils - works fine!

closed 2020-01-08 by dvdesolve