BatchDrake/suscan #33: too many arguments to function ‘SoapySDRDevice_setupStream’

nikoroxy, 2020-01-20:
While building `source.c.o` I encounter multiple warnings, concluding in an error as described in title.

OS: Ubuntu 19.10 (just installed to test SigDigger, but it happened also on Manjaro)

terminal log:
```
nik@ubuntu:~/software/suscan/build$ make
Scanning dependencies of target suscan
[  2%] Building C object CMakeFiles/suscan.dir/util/cfg.c.o
/home/nik/software/suscan/util/cfg.c: In function ‘suscan_string_to_config’:
/home/nik/software/suscan/util/cfg.c:529:29: warning: format ‘%lli’ expects argument of type ‘long long int *’, but argument 3 has type ‘uint64_t *’ {aka ‘long unsigned int *’} [-Wformat=]
  529 |         if (sscanf(val, "%lli", &int_val) < 1) {
      |                          ~~~^   ~~~~~~~~
      |                             |   |
      |                             |   uint64_t * {aka long unsigned int *}
      |                             long long int *
      |                          %li
/home/nik/software/suscan/util/cfg.c: In function ‘suscan_object_to_config’:
/home/nik/software/suscan/util/cfg.c:772:31: warning: format ‘%lli’ expects argument of type ‘long long int *’, but argument 3 has type ‘uint64_t *’ {aka ‘long unsigned int *’} [-Wformat=]
  772 |           if (sscanf(val, "%lli", &int_val) < 1) {
      |                            ~~~^   ~~~~~~~~
      |                               |   |
      |                               |   uint64_t * {aka long unsigned int *}
      |                               long long int *
      |                            %li
[  4%] Building C object CMakeFiles/suscan.dir/util/confdb.c.o
[  6%] Building C object CMakeFiles/suscan.dir/util/deserialize.c.o
[  8%] Building C object CMakeFiles/suscan.dir/util/object.c.o
[ 10%] Building C object CMakeFiles/suscan.dir/util/serialize.c.o
[ 12%] Building C object CMakeFiles/suscan.dir/util/util.c.o
[ 14%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/inspector.c.o
[ 16%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/interface.c.o
[ 18%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/params.c.o
[ 20%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/ask.c.o
[ 22%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/audio.c.o
[ 25%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/fsk.c.o
[ 27%] Building C object CMakeFiles/suscan.dir/analyzer/inspector/impl/psk.c.o
[ 29%] Building C object CMakeFiles/suscan.dir/codec/codec.c.o
[ 31%] Building C object CMakeFiles/suscan.dir/codec/codecs/diff.c.o
[ 33%] Building C object CMakeFiles/suscan.dir/analyzer/workers/channel.c.o
[ 35%] Building C object CMakeFiles/suscan.dir/analyzer/workers/wide.c.o
[ 37%] Building C object CMakeFiles/suscan.dir/analyzer/analyzer.c.o
[ 39%] Building C object CMakeFiles/suscan.dir/analyzer/bufpool.c.o
[ 41%] Building C object CMakeFiles/suscan.dir/analyzer/client.c.o
[ 43%] Building C object CMakeFiles/suscan.dir/analyzer/estimator.c.o
[ 45%] Building C object CMakeFiles/suscan.dir/analyzer/inspsched.c.o
[ 47%] Building C object CMakeFiles/suscan.dir/analyzer/insp-server.c.o
[ 50%] Building C object CMakeFiles/suscan.dir/analyzer/mq.c.o
[ 52%] Building C object CMakeFiles/suscan.dir/analyzer/msg.c.o
[ 54%] Building C object CMakeFiles/suscan.dir/analyzer/slow.c.o
[ 56%] Building C object CMakeFiles/suscan.dir/analyzer/source.c.o
/home/nik/software/suscan/analyzer/source.c: In function ‘suscan_source_open_sdr’:
/home/nik/software/suscan/analyzer/source.c:1647:7: warning: passing argument 2 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
 1647 |       &source->rx_stream,
      |       ^~~~~~~~~~~~~~~~~~
      |       |
      |       SoapySDRStream ** {aka struct SoapySDRStream **}
In file included from /home/nik/software/suscan/analyzer/source.h:30,
                 from /home/nik/software/suscan/analyzer/source.c:29:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘int’ but argument is of type ‘SoapySDRStream **’ {aka ‘struct SoapySDRStream **’}
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nik/software/suscan/analyzer/source.c:1648:7: warning: passing argument 3 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
 1648 |       SOAPY_SDR_RX,
      |       ^~~~~~~~~~~~
      |       |
      |       int
In file included from /home/nik/software/suscan/analyzer/source.h:30,
                 from /home/nik/software/suscan/analyzer/source.c:29:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const char *’ but argument is of type ‘int’
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nik/software/suscan/analyzer/source.c:1649:7: warning: passing argument 4 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Wincompatible-pointer-types]
 1649 |       SUSCAN_SOAPY_SAMPFMT,
      |       ^~~~~~~~~~~~~~~~~~~~
      |       |
      |       char *
In file included from /home/nik/software/suscan/analyzer/source.h:30,
                 from /home/nik/software/suscan/analyzer/source.c:29:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const size_t *’ {aka ‘const long unsigned int *’} but argument is of type ‘char *’
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nik/software/suscan/analyzer/source.c:1650:13: warning: passing argument 5 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
 1650 |       source->chan_array,
      |       ~~~~~~^~~~~~~~~~~~
      |             |
      |             size_t * {aka long unsigned int *}
In file included from /home/nik/software/suscan/analyzer/source.h:30,
                 from /home/nik/software/suscan/analyzer/source.c:29:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘size_t’ {aka ‘const long unsigned int’} but argument is of type ‘size_t *’ {aka ‘long unsigned int *’}
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nik/software/suscan/analyzer/source.c:1651:7: warning: passing argument 6 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
 1651 |       1,
      |       ^
      |       |
      |       int
In file included from /home/nik/software/suscan/analyzer/source.h:30,
                 from /home/nik/software/suscan/analyzer/source.c:29:
/usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const SoapySDRKwargs *’ {aka ‘const struct <anonymous> *’} but argument is of type ‘int’
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nik/software/suscan/analyzer/source.c:1645:7: error: too many arguments to function ‘SoapySDRDevice_setupStream’
 1645 |   if (SoapySDRDevice_setupStream(
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/nik/software/suscan/analyzer/source.h:30,
                 from /home/nik/software/suscan/analyzer/source.c:29:
/usr/local/include/SoapySDR/Device.h:295:31: note: declared here
  295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/suscan.dir/build.make:401: CMakeFiles/suscan.dir/analyzer/source.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/suscan.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

BatchDrake, 2020-01-21:
Yeah, I discovered yesterday that SoapySDR 0.8 broke backward-compatibility with setupStream. I've already pushed a hotfix to master, could you git-pull and try again?

nikoroxy, 2020-01-21:
Building has been fixed. 
When finished I've verified correct installation but...

```
nik@ubuntu:~$ suscan.status 
suscan.status: error while loading shared libraries: libsuscan.so: cannot open shared object file: No such file or directory
```

even if
```
nik@ubuntu:~$ sudo find / -name libsuscan.so
/usr/local/lib/libsuscan.so
/home/nik/software/suscan/build/libsuscan.so
```

BatchDrake, 2020-01-22:
That seems a library search path problem. Could you try the following?
```
$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
$ suscan.status
```

If this works, it simply means that your system is not looking for libraries in the right place. There are two possible solutions to this:

* Simply edit your `~/.bashrc` (I assume you use bash) and add `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib` at the end of the file, or
* recompile both sigutils and suscan from scratch passing `-DCMAKE_INSTALL_PREFIX=/usr` to cmake, so that libraries will be installed in `/usr/lib` instead of `/usr/local/lib`

nikoroxy, 2020-01-22:
You were right. 
I think something during `#make install` could have been messed up the path reference but I didn't payed so much attention (next time I'll redirect stdout in a file for eventual debug).

Thanks

closed 2020-01-22 by nikoroxy