BatchDrake/suscan #34: SoapySDR headers not found on macOS

mehdideveloper, 2020-01-26:
Hey
As you know I test on 2 MacBooks.
On one MacBook, I have compiled SoapySDR from source, so there's no issues.
On the other one, I have installed it via brew. On this second machine, I get error:
`/Users/mehdi/Documents/source/suscan/analyzer/source.h:30:10: fatal error: 'SoapySDR/Device.h' file not found`
Well I fixed it temporarily by setting the CPATH variable in command line.
Couldn't find any strange things in your CMakeLists

BatchDrake, 2020-01-26:
I could not reproduce this one. What version of SoapySDR are you using?

mehdideveloper, 2020-01-26:
On the one that has issues:

```
Lib Version: v0.7.2-release
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
```

BatchDrake, 2020-01-26:
Strange. pkg-config should provide the right paths for this. What does `pkg-config SoapySDR --cflags --libs` print?

mehdideveloper, 2020-01-26:
```
-L/usr/local/Cellar/soapysdr/0.7.2/lib -lSoapySDR
```
So the libraries are ok, but for some reason the headers were not found !

BatchDrake, 2020-02-08:
Changed a bunch of things in develop, any news regarding this issue?

mehdideveloper, 2020-02-09:
Hmmm. I am not sure if it was because of the change or some changes in my system, but a clean build from scratch works without problems now 👍

BatchDrake, 2020-02-09:
There was an issue with headers in general, after building in Mac I may
have fixed a bunch of issues at a time.

closed 2020-02-14 by BatchDrake