BatchDrake/suscan #36: Library not found macOS

mehdideveloper, 2020-02-03:
Hey
Fetched the new branch (feature/raw-inspector)
Clean build on macOS gives me an error when `make`ing:

```
[ 88%] Building C object CMakeFiles/suscan.dir/analyzer/spectsrcs/psd.c.o
[ 90%] Linking C shared library libsuscan.dylib
clang: error: no such file or directory: 'libsuscan.dylib'
make[2]: *** [libsuscan.dylib] Error 1
make[1]: *** [CMakeFiles/suscan.dir/all] Error 2
make: *** [all] Error 2
```

BatchDrake, 2020-02-03:
Mmh, that's odd, I didn't change anything in the build system (apart from adding a new file). Also, it is like suscan depends on itself. Is this a clean build directory? Could you attach your `CMakeLists/suscan.dir/link.txt`?

mehdideveloper, 2020-02-03:
Yes this is a clean build.
```
/Library/Developer/CommandLineTools/usr/bin/cc   -DPKGDATADIR='"/usr/local/share/suscan"' -D__FILENAME__='"$(subst /Users/user1/Downloads/source/suscan/,,$(abspath $<))"' -g -O0 -ggdb -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -dynamiclib -Wl,-headerpad_max_install_names -L/usr/local/lib -lsigutils -lsndfile -lm -lfftw3f -l -o libsuscan.dylib -install_name @rpath/libsuscan.dylib CMakeFiles/suscan.dir/util/cfg.c.o CMakeFiles/suscan.dir/util/compat.c.o CMakeFiles/suscan.dir/util/confdb.c.o CMakeFiles/suscan.dir/util/deserialize.c.o CMakeFiles/suscan.dir/util/object.c.o CMakeFiles/suscan.dir/util/serialize.c.o CMakeFiles/suscan.dir/util/util.c.o CMakeFiles/suscan.dir/analyzer/inspector/inspector.c.o CMakeFiles/suscan.dir/analyzer/inspector/interface.c.o CMakeFiles/suscan.dir/analyzer/inspector/params.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/ask.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/audio.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/fsk.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/psk.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/raw.c.o CMakeFiles/suscan.dir/codec/codec.c.o CMakeFiles/suscan.dir/codec/codecs/diff.c.o CMakeFiles/suscan.dir/analyzer/workers/channel.c.o CMakeFiles/suscan.dir/analyzer/workers/wide.c.o CMakeFiles/suscan.dir/analyzer/analyzer.c.o CMakeFiles/suscan.dir/analyzer/bufpool.c.o CMakeFiles/suscan.dir/analyzer/client.c.o CMakeFiles/suscan.dir/analyzer/estimator.c.o CMakeFiles/suscan.dir/analyzer/inspsched.c.o CMakeFiles/suscan.dir/analyzer/insp-server.c.o CMakeFiles/suscan.dir/analyzer/mq.c.o CMakeFiles/suscan.dir/analyzer/msg.c.o CMakeFiles/suscan.dir/analyzer/slow.c.o CMakeFiles/suscan.dir/analyzer/source.c.o CMakeFiles/suscan.dir/analyzer/spectsrc.c.o CMakeFiles/suscan.dir/analyzer/symbuf.c.o CMakeFiles/suscan.dir/analyzer/throttle.c.o CMakeFiles/suscan.dir/analyzer/worker.c.o CMakeFiles/suscan.dir/analyzer/estimators/fac.c.o CMakeFiles/suscan.dir/analyzer/estimators/nonlinear.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/cyclo.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/fmcyclo.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/fmspect.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/pmspect.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/timediff.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/exp-2.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/exp-4.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/exp-8.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/psd.c.o   -L/Users/user1/Downloads/source/suscan/build  -L/usr/local/Cellar/libsndfile/1.0.28/lib  -L/usr/local/Cellar/fftw/3.3.8_1/lib  -L/usr/local/lib  -Wl,-rpath,/Users/user1/Downloads/source/suscan/build -Wl,-rpath,/usr/local/Cellar/libsndfile/1.0.28/lib -Wl,-rpath,/usr/local/Cellar/fftw/3.3.8_1/lib -Wl,-rpath,/usr/local/lib -lm -lsigutils -lsndfile -lm -lfftw3f -lsndfile -lsndfile -lfftw3f -lSoapySDR -lxml2 -lm -lsigutils -lSoapySDR -lxml2 
```

mehdideveloper, 2020-02-03:
Yes it's a clean build.
[link.txt](https://github.com/BatchDrake/suscan/files/4148473/link.txt)

BatchDrake, 2020-02-04:
Can't reproduce in MacOS Catalina :/ Here's my link.txt, just in case you want to compare:

```
/Library/Developer/CommandLineTools/usr/bin/cc   -DPKGDATADIR='"/usr/local/share/suscan"' -D__FILENAME__='"$(subst /Users/waldo/Documents/Development/suscan/,,$(abspath $<))"' -DHAVE_VOLK=1 -g -O0 -ggdb -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -dynamiclib -Wl,-headerpad_max_install_names -L/usr/local/lib -lsigutils -lsndfile -lm -lfftw3f -lvolk -o libsuscan.dylib -install_name @rpath/libsuscan.dylib CMakeFiles/suscan.dir/util/cfg.c.o CMakeFiles/suscan.dir/util/compat.c.o CMakeFiles/suscan.dir/util/confdb.c.o CMakeFiles/suscan.dir/util/deserialize.c.o CMakeFiles/suscan.dir/util/object.c.o CMakeFiles/suscan.dir/util/serialize.c.o CMakeFiles/suscan.dir/util/util.c.o CMakeFiles/suscan.dir/analyzer/inspector/inspector.c.o CMakeFiles/suscan.dir/analyzer/inspector/interface.c.o CMakeFiles/suscan.dir/analyzer/inspector/params.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/ask.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/audio.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/fsk.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/psk.c.o CMakeFiles/suscan.dir/analyzer/inspector/impl/raw.c.o CMakeFiles/suscan.dir/codec/codec.c.o CMakeFiles/suscan.dir/codec/codecs/diff.c.o CMakeFiles/suscan.dir/analyzer/workers/channel.c.o CMakeFiles/suscan.dir/analyzer/workers/wide.c.o CMakeFiles/suscan.dir/analyzer/analyzer.c.o CMakeFiles/suscan.dir/analyzer/bufpool.c.o CMakeFiles/suscan.dir/analyzer/client.c.o CMakeFiles/suscan.dir/analyzer/estimator.c.o CMakeFiles/suscan.dir/analyzer/inspsched.c.o CMakeFiles/suscan.dir/analyzer/insp-server.c.o CMakeFiles/suscan.dir/analyzer/mq.c.o CMakeFiles/suscan.dir/analyzer/msg.c.o CMakeFiles/suscan.dir/analyzer/slow.c.o CMakeFiles/suscan.dir/analyzer/source.c.o CMakeFiles/suscan.dir/analyzer/spectsrc.c.o CMakeFiles/suscan.dir/analyzer/symbuf.c.o CMakeFiles/suscan.dir/analyzer/throttle.c.o CMakeFiles/suscan.dir/analyzer/worker.c.o CMakeFiles/suscan.dir/analyzer/estimators/fac.c.o CMakeFiles/suscan.dir/analyzer/estimators/nonlinear.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/cyclo.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/fmcyclo.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/fmspect.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/pmspect.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/timediff.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/exp-2.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/exp-4.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/exp-8.c.o CMakeFiles/suscan.dir/analyzer/spectsrcs/psd.c.o  -L/Users/waldo/Documents/Development/suscan/build  -L/usr/local/Cellar/libsndfile/1.0.28/lib  -L/usr/local/Cellar/fftw/3.3.8_1/lib  -L/usr/local/Cellar/soapysdr/0.7.1/lib  -L/usr/local/lib -Wl,-rpath,/Users/waldo/Documents/Development/suscan/build -Wl,-rpath,/usr/local/Cellar/libsndfile/1.0.28/lib -Wl,-rpath,/usr/local/Cellar/fftw/3.3.8_1/lib -Wl,-rpath,/usr/local/Cellar/soapysdr/0.7.1/lib -Wl,-rpath,/usr/local/lib -lm -lsigutils -lsndfile -lm -lfftw3f -lvolk -lsndfile -lsndfile -lfftw3f -lSoapySDR -lxml2 -lm -lsigutils -lvolk -lSoapySDR -lxml2 
```

mehdideveloper, 2020-02-04:
As you can see, it's because of `volk`. Because I don't have `volk`, an empty `-l` is added there which breaks the linker. It happens on clean builds, because I guess it's introduced by the Cmake checker

BatchDrake, 2020-02-08:
Just updated everything, problem was originated in sigutils, pull from develop and try again.

mehdideveloper, 2020-02-09:
Thanks. This fixed the issue for me 👍

closed 2020-02-09 by BatchDrake