BatchDrake/suscan #37: clock_gettime on macOS

mehdideveloper, 2020-02-14:
Hey
clock_gettime is not implemented on macOS (neither its related constants )
Look at this for example: https://stackoverflow.com/questions/5167269/clock-gettime-alternative-in-mac-os-x

mehdideveloper, 2020-02-14:
And I'm not sure what you think about this:

https://github.com/envelope-project/laik/commit/86c180e1c16fffa0761a785dad315a7d524e053b

BatchDrake, 2020-02-14:
We've been using `clock_gettime` in MacOS for a whie now, what does not exist is `CLOCK_MONOTONIC_COARSE`. We may need to fallback to `CLOCK_MONOTONIC`

BatchDrake, 2020-02-14:
Also, maybe it is a good idea to rely on Mach functions directly, as described in https://opensource.apple.com/source/Libc/Libc-1158.1.2/gen/clock_gettime.c.auto.html

BatchDrake, 2020-02-14:
Also, is `mach_boottime_usec` documented anywhere?

mehdideveloper, 2020-02-14:
Ok. I was wrong. It was not available before OS X 10.12.
I checked `time.h` and it actually is there.
Regarding `CLOCK_MONOTONIC_COARSE`: 
I searched my `time.h` and this is what I have:
```
 _CLOCK_MONOTONIC __CLOCK_AVAILABILITY = 6,
#define CLOCK_MONOTONIC _CLOCK_MONOTONIC
#if !defined(_POSIX_C_SOURCE) || defined(_DARWIN_C_SOURCE)
_CLOCK_MONOTONIC_RAW __CLOCK_AVAILABILITY = 4,
#define CLOCK_MONOTONIC_RAW _CLOCK_MONOTONIC_RAW
_CLOCK_MONOTONIC_RAW_APPROX __CLOCK_AVAILABILITY = 5,
#define CLOCK_MONOTONIC_RAW_APPROX _CLOCK_MONOTONIC_RAW_APPROX
```

mehdideveloper, 2020-02-14:
Where does `mach_boottime_usec` come from? I can't find it in any files in my `include`

sudeepan, 2022-04-19:
Is there a `mach` equivalent of `CLOCK_MONOTONIC_COARSE` ?

sudeepan, 2022-04-20:
> Ok. I was wrong. It was not available before OS X 10.12.
> I checked `time.h` and it actually is there.
> Regarding `CLOCK_MONOTONIC_COARSE`:
> I searched my `time.h` and this is what I have:
> ``` [...]

Is there a `mach` equivalent of `CLOCK_MONOTONIC_COARSE` ?

sudeepan, 2022-04-20:
> We've been using `clock_gettime` in MacOS for a whie now, what does not exist is `CLOCK_MONOTONIC_COARSE`. We may need to fallback to `CLOCK_MONOTONIC`

Is there a `mach` equivalent of `CLOCK_MONOTONIC_COARSE` ?

closed 2023-04-12 by BatchDrake