BatchDrake/suscan #49: cmake fail linux can't find sigutils after installing sigutils

airhamer, 2020-06-15:
Just installed sigutils.
I am afraid I don't understand how packages are found and verified by cmake.  But sigutils compiled and installed successfully.  The install prefix was /usr   I tried it in /usr/local as well, but neither install prefix resulted in successful cmake for suscan.

cmake -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 8.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'sigutils>=0.1'
--   No package 'sigutils' found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:467 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:647 (_pkg_check_modules_internal)
  CMakeLists.txt:28 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/steve/Downloads/suscan/build/CMakeFiles/CMakeOutput.log".
See also "/home/steve/Downloads/suscan/build/CMakeFiles/CMakeError.log".
[CMakeError.log](https://github.com/BatchDrake/suscan/files/4783191/CMakeError.log)
[CMakeOutput.log](https://github.com/BatchDrake/suscan/files/4783192/CMakeOutput.log)

BatchDrake, 2020-06-17:
Hi,

Could you paste the output of `pkg-config --variable pc_path pkg-config` here? Also, what messages do you see after running `sudo make install` in sigutils?

Cheers,

BatchDrake, 2020-07-22:
Since this has been opened for a while with no responses, I'm closing it for now. Feel free to reopen it if you still have issues building suscan.

closed 2020-07-22 by BatchDrake

probinso, 2021-05-30:
I ran into this same problem.

```
$ cat /etc/ld.so.conf.d/local.conf
/usr/local/lib
/usr/local/lib64/
``` 

```
$ pkg-config --variable pc_path pkg-config
/usr/lib64/pkgconfig:/usr/share/pkgconfig
```

```
$ ldconfig -p | grep sigutil
	libsigutils.so (libc6,x86-64) => /usr/local/lib/libsigutils.so
```

BatchDrake, 2021-05-30:
Hi @probinso

According to this output:

```
$ pkg-config --variable pc_path pkg-config
/usr/lib64/pkgconfig:/usr/share/pkgconfig
```

It looks like you have installed sigutils under `/usr/local`, but `pkg-config` is looking for packages in `/usr/lib64` and `/usr/share`. You have to include `/usr/local/lib/pkgconfig` in the `PKG_CONFIG_PATH` before compiling suscan. Please run in the same terminal you are using to build suscan:

```
$ export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:$PKG_CONFIG_PATH
```

then try to build suscan again, and tell me if you still get errors.

probinso, 2021-05-30:
I got an error with `SoapySDR` missing, but it turns out that I hadn't installed `SoapySDR-devel`