BatchDrake/suscan #50: suscan doesn't actually use volk?

FFY00, 2020-07-21:
Hi, when packing this for arch I noticed the following warning. Looks like suscan isn't actually using volk, if that's the case, the dependency can be removed. Could you investigate? Thanks :grin.

```
suscan W: Unused shared library '/usr/lib/libvolk.so.2.3' by file ('usr/lib/libsuscan.so')
```

BatchDrake, 2020-07-22:
You are right, Volk is used in suscan _through_ sigutils, and the only direct use of Volk comes from certain macros in ```ncqo.h```. I'll remove this dependency ASAP.

BatchDrake, 2020-07-22:
This ended up being a sigutils issue, related to the corresponding .pc file referring to unnecesary libs (volk and fftw3f). Changes are in develop and scheduled for the next release. SigDigger has been changed too in order to rely on fftw3f explicitly.

BatchDrake, 2020-07-22:
I forgot to mention that the changes are in the `develop` branch of all projects.

FFY00, 2020-07-22:
Hum, okay. I'll wait for a release, thanks!

BatchDrake, 2021-06-30:
FYI I am already working on 0.2.0. I'm marking this as fixed as changes are already in develop.

closed 2021-06-30 by BatchDrake

FFY00, 2021-06-30:
Thanks :blush: