BatchDrake/suscan #54: Suscli radio profile setting not reconiz

alphafox02, 2021-09-12:
Before I forget. With the latest Suscli dev branch, I'm unable to change the radio profile using the following commands

Suscli radio profile 2 (replace 2 with any profile index or name)

Initially I was stuck with whatever UI Profile. I then deleted the Suscan sources file. Used Suscli makeprof to create new profiles which resulted in the Default, Airspy, and an IP/port profile.

Now when I start Suscli radio, no matter which profile I pick, it's the Airspy device so at least I can use it. However, I think something is preventing the user from changing profiles.

alphafox02, 2021-09-12:
After deleting the profiles and trying again, it does seem to be working as intended. I'll keep testing and close this issue if that's the case.

BatchDrake, 2021-09-12:
Hi,

You are missing the = between `profile` and `2`

```
$ suscli radio profile=2
```

alphafox02, 2021-09-12:
My mistake, that was it. Profiles are selectable.

closed 2021-09-12 by alphafox02