BatchDrake/suscan #59: suscli exceptions on Mac OSX

rabarar, 2022-03-31:
Any idea why these exceptions print when running `suscli devices`
```
./suscli devices      
[e] suscan_bundle_get_resource_path:45: exception in "dir_url = CFBundleCopyResourceURL( main_bundle, relpath, ((void*)0), ((void*)0) )" (build/CMakeFiles/suscan.dir/compiler_depend.ts:45)
[e] suscan_bundle_get_resource_path:45: exception in "dir_url = CFBundleCopyResourceURL( main_bundle, relpath, ((void*)0), ((void*)0) )" (build/CMakeFiles/suscan.dir/compiler_depend.ts:45)
```

andreascla, 2022-05-11:
I have the same issue, in addition I can see that the "Dummy device" is unavailable if that helps

```
% suscli devices
11 May 2022 - 08:50:40   exception in "dir_url = CFBundleCopyResourceURL( main_bundle, relpath, ((void*)0), ((void*)0) )" (build/CMakeFiles/suscan.dir/compiler_depend.ts:45)
11 May 2022 - 08:50:40   exception in "dir_url = CFBundleCopyResourceURL( main_bundle, relpath, ((void*)0), ((void*)0) )" (build/CMakeFiles/suscan.dir/compiler_depend.ts:45)
 ndx Device name                              Driver   Interface Availability
------------------------------------------------------------------------------
[ 0] Dummy device                             null     local     unavailable
```

I have built this from scratch on two different Mac Book M1, but the issue is only present on one of them so I suspect there is an issue with one of the dependencies or build environment.

andreascla, 2022-05-11:
The issue is definitely related to SoapySDR. I rebuilt it from scratch (as opposed to install it from brew) and the suscan library now works without error messages:

```
% suscli devices
 ndx Device name                              Driver   Interface Availability
------------------------------------------------------------------------------
[ 0] Dummy device                             null     local     unavailable
```

BatchDrake, 2022-05-13:
Nah, it was actually an improper exception handling from my part. By the way, what is the output when you plug any SDR (e.g. HackRF) and run suscli devices again?

andreascla, 2022-05-18:
This is what it looks like with a USRP B205 connected:

```
% suscli devices
 ndx Device name                              Driver   Interface Availability
------------------------------------------------------------------------------
[ 0] Dummy device                             null     local     unavailable
[ 1] uhd (B205mini 31B66CC)                   uhd      local     available
```

Not sure why the "Dummy device" is "unavailable" though, but not sure if it matters.

BatchDrake, 2022-05-18:
Don't worry about that. The dummy device, although seen by Suscan, it is not instantiable by SoapySDR (hence its unavailability). Think of it as some kind of placeholder when there are no connected devices whatsoever.

closed 2023-04-12 by BatchDrake