BatchDrake/suscan #61: Build warnings around yaml on FreeBSD

jeffpc, 2022-04-11:
I don't feel like figuring out the yaml macros, so here's a bug report ;)

```
/home/jeffpc/src/oss/dsp/suscan/util/serialize-yaml.c:97:7: warning: implicit conversion from enumeration type 'enum yaml_mapping_style_e' to different enumeration type 'yaml_sequence_style_t' (aka 'enum yaml_sequence_style_e') [-Wenum-conversion]
      SUSCAN_YAML_SEQ_START();
      ^~~~~~~~~~~~~~~~~~~~~~~
/home/jeffpc/src/oss/dsp/suscan/util/serialize-yaml.c:58:5: note: expanded from macro 'SUSCAN_YAML_SEQ_START'
    YAML_BLOCK_MAPPING_STYLE)
    ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/jeffpc/src/oss/dsp/suscan/util/serialize-yaml.c:36:9: note: expanded from macro 'SUSCAN_YAML_EMIT'
      ##arg)));                                                   \
      ~~^~~~~
/home/jeffpc/bin/dsp/include/sigutils/util/util.h:52:29: note: expanded from macro 'JOIN'
#define JOIN(a, b) _JOIN(a, b)
                   ~~~~~~~~~^~
/home/jeffpc/bin/dsp/include/sigutils/util/util.h:51:26: note: expanded from macro '_JOIN'
#define _JOIN(a, b) a ## b
                    ~~~~~^
/home/jeffpc/bin/dsp/include/sigutils/sigutils/defs.h:130:36: note: expanded from macro 'SU_TRY'
#define SU_TRY(expr)   SU_TRYCATCH(expr, goto done)
                                   ^~~~
/home/jeffpc/bin/dsp/include/sigutils/sigutils/defs.h:120:9: note: expanded from macro 'SU_TRYCATCH'
  if (!(expr)) {                        \
        ^~~~
/home/jeffpc/src/oss/dsp/suscan/util/serialize-yaml.c:174:3: warning: implicit conversion from enumeration type 'enum yaml_mapping_style_e' to different enumeration type 'yaml_sequence_style_t' (aka 'enum yaml_sequence_style_e') [-Wenum-conversion]
  SUSCAN_YAML_SEQ_START();
  ^~~~~~~~~~~~~~~~~~~~~~~
/home/jeffpc/src/oss/dsp/suscan/util/serialize-yaml.c:58:5: note: expanded from macro 'SUSCAN_YAML_SEQ_START'
    YAML_BLOCK_MAPPING_STYLE)
    ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/jeffpc/src/oss/dsp/suscan/util/serialize-yaml.c:36:9: note: expanded from macro 'SUSCAN_YAML_EMIT'
      ##arg)));                                                   \
      ~~^~~~~
/home/jeffpc/bin/dsp/include/sigutils/util/util.h:52:29: note: expanded from macro 'JOIN'
#define JOIN(a, b) _JOIN(a, b)
                   ~~~~~~~~~^~
/home/jeffpc/bin/dsp/include/sigutils/util/util.h:51:26: note: expanded from macro '_JOIN'
#define _JOIN(a, b) a ## b
                    ~~~~~^
/home/jeffpc/bin/dsp/include/sigutils/sigutils/defs.h:130:36: note: expanded from macro 'SU_TRY'
#define SU_TRY(expr)   SU_TRYCATCH(expr, goto done)
                                   ^~~~
/home/jeffpc/bin/dsp/include/sigutils/sigutils/defs.h:120:9: note: expanded from macro 'SU_TRYCATCH'
  if (!(expr)) {                        \
        ^~~~
2 warnings generated.
```

This is with the default FreeBSD 13 compiler:

```
$ cc -v
FreeBSD clang version 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c3fe)
Target: x86_64-unknown-freebsd13.0
Thread model: posix
InstalledDir: /usr/bin
```

BatchDrake, 2022-04-11:
Thanks, looking into it now.

BatchDrake, 2022-04-11:
It should be fixed in commit ce710db now, could you try to build again?

jeffpc, 2022-04-11:
Yep, this change (plus my socket include changes) result in a clean build with zero warnings.  Thanks.

closed 2022-04-11 by BatchDrake