BatchDrake/suscan #62: suscan crashing with "invalid opcode" error

alexsystems55, 2023-01-14:
Hi!
I have both SigDigger and suscan.status crashing on a fresh installed Mint Linux 21.1 with following error:
```
Jan 15 00:05:55 kernel: [  498.948942] traps: SigDigger::Init[2118] trap invalid opcode ip:7f6ef8001660 sp:7f6edf7fce00 error:0 in librte_net.so.22.0[7f6ef8001000+3000]

Jan 15 00:09:42 kernel: [  725.825914] traps: suscan.status[2629] trap invalid opcode ip:7f967ec6e660 sp:7ffd5e200040 error:0 in librte_net.so.22.0[7f967ec6e000+3000]
```
Few days ago everything worked fine on Ubuntu Studio 20.04 on the same PC. Any ideas what went wrong?

BatchDrake, 2023-01-14:
Hi,

I'd say this error is occuring not in SigDigger / suscan but in something else, `librte_net.so.22.0`, which I do not depend on directly. In any case, invalid opcode errors are rare, and point to system or build errors (like mixing up architectures during an install). 

What does `uname -a` print? How are you running SigDigger? Is this a custom build or are you running an AppImage?

alexsystems55, 2023-01-14:
`Linux alexsystem 5.15.0-58-generic #64-Ubuntu SMP Thu Jan 5 11:43:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`
SigDigger was built from sources, following the readme, and run from terminal by typing `SigDigger`.

I didn't notice the AppImage option before, looks like it works fine. Thanks for pointing it out.

alexsystems55, 2023-01-14:
Looks like not suscan issue.

closed 2023-01-14 by alexsystems55