BatchDrake/suscan #79: Build issues on arm64 ubuntu 18.04

mmoscony, 2023-09-07:
Trying to install and i get the following error when i run make:
`
[ 83%] Linking C executable suscan.status
libsuscan.so.0.3.0: undefined reference to `mpg123_format'
libsuscan.so.0.3.0: undefined reference to `id3tag_genre_list'
libsuscan.so.0.3.0: undefined reference to `vorbis_analysis_headerout'
libsuscan.so.0.3.0: undefined reference to `lame_set_out_samplerate'
libsuscan.so.0.3.0: undefined reference to `ogg_stream_packetin'
libsuscan.so.0.3.0: undefined reference to `lame_encode_buffer_float'
libsuscan.so.0.3.0: undefined reference to `lame_encode_buffer'
libsuscan.so.0.3.0: undefined reference to `lame_get_encoder_delay'
libsuscan.so.0.3.0: undefined reference to `mpg123_new'
libsuscan.so.0.3.0: undefined reference to `lame_set_VBR_mean_bitrate_kbps'
libsuscan.so.0.3.0: undefined reference to `FLAC__metadata_object_vorbiscomment_find_entry_from'
libsuscan.so.0.3.0: undefined reference to `FLAC__StreamEncoderInitStatusString'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_decoder_process_until_end_of_metadata'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_decoder_seek_absolute'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_decoder_destroy'
libsuscan.so.0.3.0: undefined reference to `ogg_page_eos'
libsuscan.so.0.3.0: undefined reference to `mpg123_param'
libsuscan.so.0.3.0: undefined reference to `ogg_packet_clear'
libsuscan.so.0.3.0: undefined reference to `mpg123_plain_strerror'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_encode_float'
libsuscan.so.0.3.0: undefined reference to `lame_encode_buffer_interleaved_ieee_double'
libsuscan.so.0.3.0: undefined reference to `lame_set_write_id3tag_automatic'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_decoder_create'
libsuscan.so.0.3.0: undefined reference to `vorbis_synthesis_headerin'
libsuscan.so.0.3.0: undefined reference to `vorbis_synthesis_trackonly'
libsuscan.so.0.3.0: undefined reference to `lame_set_brate'
libsuscan.so.0.3.0: undefined reference to `mpg123_open_handle'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_encoder_ctl'
libsuscan.so.0.3.0: undefined reference to `vorbis_encode_init_vbr'
libsuscan.so.0.3.0: undefined reference to `FLAC__metadata_object_new'
libsuscan.so.0.3.0: undefined reference to `ogg_sync_init'
libsuscan.so.0.3.0: undefined reference to `FLAC__metadata_object_delete'
libsuscan.so.0.3.0: undefined reference to `ogg_sync_reset'
libsuscan.so.0.3.0: undefined reference to `vorbis_analysis_blockout'
libsuscan.so.0.3.0: undefined reference to `vorbis_info_clear'
libsuscan.so.0.3.0: undefined reference to `vorbis_analysis_buffer'
libsuscan.so.0.3.0: undefined reference to `lame_get_version'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_encoder_destroy'
libsuscan.so.0.3.0: undefined reference to `lame_encode_buffer_interleaved_ieee_float'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_surround_encoder_create'
libsuscan.so.0.3.0: undefined reference to `mpg123_read'
libsuscan.so.0.3.0: undefined reference to `lame_set_bWriteVbrTag'
libsuscan.so.0.3.0: undefined reference to `mpg123_length'
libsuscan.so.0.3.0: undefined reference to `vorbis_synthesis'
libsuscan.so.0.3.0: undefined reference to `lame_get_VBR_mean_bitrate_kbps'
libsuscan.so.0.3.0: undefined reference to `opus_multistream_decode_float'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_decoder_set_metadata_respond_all'
libsuscan.so.0.3.0: undefined reference to `mpg123_replace_reader_handle'
libsuscan.so.0.3.0: undefined reference to `FLAC__StreamDecoderErrorStatusString'
libsuscan.so.0.3.0: undefined reference to `id3tag_set_track'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_encoder_finish'
libsuscan.so.0.3.0: undefined reference to `lame_get_brate'
libsuscan.so.0.3.0: undefined reference to `vorbis_comment_query'
libsuscan.so.0.3.0: undefined reference to `ogg_sync_pageseek'
libsuscan.so.0.3.0: undefined reference to `FLAC__StreamDecoderStateString'
libsuscan.so.0.3.0: undefined reference to `vorbis_block_clear'
libsuscan.so.0.3.0: undefined reference to `lame_close'
libsuscan.so.0.3.0: undefined reference to `lame_set_VBR'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_decoder_finish'
libsuscan.so.0.3.0: undefined reference to `mpg123_getformat'
libsuscan.so.0.3.0: undefined reference to `lame_get_VBR'
libsuscan.so.0.3.0: undefined reference to `id3tag_set_genre'
libsuscan.so.0.3.0: undefined reference to `vorbis_synthesis_blockin'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_encoder_process_interleaved'
libsuscan.so.0.3.0: undefined reference to `ogg_page_serialno'
libsuscan.so.0.3.0: undefined reference to `vorbis_analysis'
libsuscan.so.0.3.0: undefined reference to `lame_encode_buffer_ieee_double'
[... 62 lines cut ...]
libsuscan.so.0.3.0: undefined reference to `opus_multistream_encoder_create'
libsuscan.so.0.3.0: undefined reference to `vorbis_comment_init'
libsuscan.so.0.3.0: undefined reference to `vorbis_comment_clear'
libsuscan.so.0.3.0: undefined reference to `lame_init_params'
libsuscan.so.0.3.0: undefined reference to `ogg_stream_clear'
libsuscan.so.0.3.0: undefined reference to `ogg_page_bos'
libsuscan.so.0.3.0: undefined reference to `id3tag_init'
libsuscan.so.0.3.0: undefined reference to `vorbis_bitrate_addblock'
libsuscan.so.0.3.0: undefined reference to `ogg_page_packets'
libsuscan.so.0.3.0: undefined reference to `ogg_stream_init'
libsuscan.so.0.3.0: undefined reference to `mpg123_info'
libsuscan.so.0.3.0: undefined reference to `vorbis_comment_add_tag'
libsuscan.so.0.3.0: undefined reference to `lame_encode_buffer_int'
libsuscan.so.0.3.0: undefined reference to `mpg123_scan'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_decoder_get_state'
libsuscan.so.0.3.0: undefined reference to `vorbis_bitrate_flushpacket'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_decoder_init_stream'
libsuscan.so.0.3.0: undefined reference to `vorbis_analysis_wrote'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_encoder_delete'
libsuscan.so.0.3.0: undefined reference to `mpg123_close'
libsuscan.so.0.3.0: undefined reference to `vorbis_info_init'
libsuscan.so.0.3.0: undefined reference to `FLAC__stream_encoder_set_bits_per_sample'
libsuscan.so.0.3.0: undefined reference to `ogg_sync_clear'
libsuscan.so.0.3.0: undefined reference to `lame_get_mode'
`

BatchDrake, 2023-09-07:
Looks like an OS issue. I do not use any of those symbols, so my bet is that they come from a broken install of libsndfile. Could it be that you built libsndfile from source?

Cheers,

mmoscony, 2023-09-07:
you were correct i installed using apt-get instead and got through that, thanks!

closed 2023-09-07 by mmoscony