BatchDrake/suscan #80: Saving (truly) raw IQ samples

jeffpc, 2023-09-13:
It is unfortunate that every sample suscan/SigDigger saves is a cf32 since most SDRs have nowhere near the resolution. For example, when I use SigDigger to save IQ samples from an RTL-SDR device, the resulting file is 4x the size it would be if I used rtl_sdr/rtl_fm -M raw/etc. even though it doesn't actually contain any additional information.  (Fancier 9..16-bit SDRs may suffer "only" 2x growth.)

So, it would be nice if there was an option to save the original SDR samples instead of the lightly processed (gain only?) cf32 samples.

I looked at the cf32 data in a recording I made, and I saw more than 256 unique floats, so the saved samples appear to be lightly processed.

1. This size factor is ok for short-lived recordings, but longer-term archiving gets much costlier.
2. Any kind of processing done before the sample is written is hard or impossible to undo later on.  For example, any bugs or misconfiguration in the processing make the recording much less useful.  (This is similar to saving raw files in digital photography vs. even lightly processed non-raw files.)

BatchDrake, 2023-09-13:
The original SDR samples are read as complex float 32 directly (this is done by SoapySDR, perhaps that's the extra processing you see), and Suscan is therefore a blind of the underlying precision of the samples. 

Perhaps this could be done by providing an option to save samples with arbitrary precision (1 bit, 8 bit, 16 bit...), but this should be done in SigDigger (which is the tool that actually saves the samples). It should not be too difficult actually ([see code of SourceWidget](https://github.com/BatchDrake/SigDigger/blob/4a4dabf64ee3506b0d5ccf744dda1041029970c5/Default/Source/SourceWidget.cpp#L1081)).

jeffpc, 2023-09-14:
> The original SDR samples are read as complex float 32 directly (this is done by SoapySDR, perhaps that's the extra processing you see), and Suscan is therefore a blind of the underlying precision of the samples.

I was afraid of that.  It would take a bit of rearchitecting, but if SoapySDR supports returning the raw samples (I don't know if it does), suscan could do the conversion to cf32 itself but also keep the original samples in an adjacent buffer.  In a way, outputting two streams.  SigDigger could then save the original one if desired.  This is likely not worth it for this feature - but I recall reading that you want to support multiple streams for multi-antenna setups so maybe this could piggyback on that work somehow.

> Perhaps this could be done by providing an option to save samples with arbitrary precision (1 bit, 8 bit, 16 bit...), 

That's an interesting idea and probably a good one in general.  While this still has the problem of not saving the truly raw data, at least it lets the user decide how much space to use.

> but this should be done in SigDigger (which is the tool that actually saves the samples).

I tried to be clever and file the bug in the right place and got it exactly backwards :)