BatchDrake/suscan #82: reserved identifier violation

elfring, 2023-11-24:
I would like to point out that identifiers like “[`_CLI_CMDS_H`](https://github.com/BatchDrake/suscan/blob/d6bfcce15bd29967643cdf60556dc616372d8f72/cli/cmds.h#L20-L21 "Update candidate")” and “[`_MAIN_INCLUDE_H`](https://github.com/BatchDrake/suscan/blob/d6bfcce15bd29967643cdf60556dc616372d8f72/src/suscan.h#L6-L7 "Another update candidate")” [do not fit](https://wiki.sei.cmu.edu/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier#DCL51CPP.Donotdeclareordefineareservedidentifier-NoncompliantCodeExample%28HeaderGuard%29 "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

BatchDrake, 2023-11-24:
Not much, actually. It is clear that you ran some kind of unattended tool on my repository that does not work particularly well (this project is written in C and not in C++) and this issue has been opened automatically.

While I welcome the usage of static analysis tools in my projects, I have a strong policy against unsupervised algorithms driving the changes in my code.

If you truly find these macros to be problematic in some systems, please clarify why so we can discuss a (human supervised) solution to it.

elfring, 2023-11-24:
:thought_balloon: How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

BatchDrake, 2023-11-24:
It depends. Are you a human or an LLM?

elfring, 2023-11-24:
>Are you a human

Yes.

>or an LLM?

No.

BatchDrake, 2023-11-24:
Okay, in that case:

* How many changes do we need to prevent UB?
* What operating system do you have?
* How are you going to test these changes?

elfring, 2023-11-24:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html "Checking the application of reserved identifiers")?

BatchDrake, 2023-11-24:
Sorry, you don't seem like a human to me anymore. Please open an issue if you truly find a problem with the code.

closed 2023-11-24 by BatchDrake

elfring, 2023-11-24:
>Please open an issue if you truly find a problem with the code.

:eyes: I obviously did that.