BatchDrake/suscan #83: Remove unnecessary null pointer checks

elfring, 2023-11-24:
[An extra null pointer check is not needed in functions](https://stackoverflow.com/questions/18775608/free-a-null-pointer-anyway-or-check-first "Free a null pointer anyway or check first?") like the following.
- [suscan_message_destroy](https://github.com/BatchDrake/suscan/blob/d6bfcce15bd29967643cdf60556dc616372d8f72/src/lib.c#L128-L132)
- [suscan_source_device_fix_airspy_rates](https://github.com/BatchDrake/suscan/blob/d6bfcce15bd29967643cdf60556dc616372d8f72/analyzer/kludges.c#L60)

BatchDrake, 2023-11-24:
Nope. I want these NULL checks so that I can replace `free()` by a custom implementation that aborts on an invalid free.

Please restrain yourself from running unsupervised code analysis tools that open issues automatically without the necessary context.

elfring, 2023-11-24:
:thought_balloon: Will your custom implementation adhere to documented programming interface requirements for null pointer tolerance?

BatchDrake, 2023-11-24:
Nope, as freeing a null pointer would be treated as an error. In my code I maintain a policy in which the number of frees must match the number of successful allocations. This is a consistency condition I impose in my projects.

elfring, 2023-11-24:
>…, as freeing a null pointer would be treated as an error.

Do you disagree to standardised API properties?

BatchDrake, 2023-11-24:
Yes.

closed 2023-11-25 by BatchDrake