BatchDrake/suscan #84: Completion of error handling

elfring, 2023-11-24:
Would you like to add more error handling for return values from functions like the following?
* [fprintf](https://pubs.opengroup.org/onlinepubs/9699919799/functions/fprintf.html "Print formatted output.") ⇒ [suscli_matlab_datasaver_close_cb](https://github.com/BatchDrake/suscan/blob/d6bfcce15bd29967643cdf60556dc616372d8f72/cli/datasavers/matlab.c#L118)
* [pthread_mutex_init](https://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_init.html "Initialize a mutex.") ⇒ [suscan_local_analyzer_source_init](https://github.com/BatchDrake/suscan/blob/d6bfcce15bd29967643cdf60556dc616372d8f72/analyzer/impl/local.c#L513)

BatchDrake, 2023-11-24:
Okay, how would you do it?

elfring, 2023-11-25:
* :thinking: How do you think about to improve [static source code analysis](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2C_C.2B.2B "Tools for code analysis") also for this software?
* :crystal_ball: How will chances evolve to benefit any more also from [the means of aspect-oriented software development](https://en.wikipedia.org/wiki/Aspect-oriented_programming "Documentation")?

closed 2023-11-25 by BatchDrake