BatchDrake/suscan #93: Problems with building -no package SoapySDR found

kenwawa01, 2024-06-17:
Hi!
In theory simple building error after cmake ..:
```

-- Checking for module 'SoapySDR>=0.5.0'
--   No package 'SoapySDR' found
CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  CMakeLists.txt:74 (pkg_check_modules)
```

It suggest that libraries are not installed. However I tried installing it and it seems I have everything:

`sudo apt-get install soapysdr-tools soapysdr-module-lms7 python3-soapysdr`

When doing:

`python3 -c "import sys; print(sys.path)"`

I get:

`['', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/usr/local/lib/python3.10/dist-packages', '/usr/lib/python3/dist-packages']`

And when manually searching on hdd I have many files with SoapySDR in name, in particular

```
SoapySDR.py in /usr/lib/python3/dist-packages

libSoapySDR.so.0.8 in /usr/lib/x86_64-linux-gnu
libSoapySDR.so.0.8.1 in /usr/lib/x86_64-linux-gnu
```

Any idea why I have this compile error?

BatchDrake, 2024-06-17:
Hi,

This is happening because you have installed the compiled library, but not the build files. You have to install `libsoapysdr-dev`

If you want to install all required packages in a single command, just run:

```bash
sudo apt-get install libsndfile1-dev libjson-c-dev libvolk2-dev libfftw3-dev soapysdr-module-all libsoapysdr-dev libxml2-dev portaudio19-dev
```

kenwawa01, 2024-06-17:
Perfect! Now I've just build! Thank you!