BatchDrake/suscan #97: Missing install step in Ubuntu

huzeyfe-erkek, 2024-10-20:
Hello, In Ubuntu `sudo ldconfig` should be called after `make install`. I kindly suggest updating README.md

BatchDrake, 2024-10-26:
Hi, and sorry for the late reply.

This is actually a build issue. I am surprised this is not triggered by the `make install` step. I am looking into it right now, because the user should not do any extra steps after build (there are enough stuff to build already!)

Thanks for reporting!

BatchDrake, 2024-10-27:
Hi again,

Just wanted to let you know I modified both suscan and sigutils yesterday so ldconfig is run in the install step. It should work out of the box now.

Cheers,

huzeyfe-erkek, 2024-10-28:
Hi, I tested the latest commit. The "ldconfig" command no longer needs to be called explicitly. Thanks.

closed 2024-10-28 by huzeyfe-erkek