BatchDrake/suscan #98: MacOS Build Issue

IanStewart2001, 2025-01-10:
When I am following the PDF instructions, and I run "make" from the build directory, I get a bunch of linker errors. Below is a snippet. 

Undefined symbols for architecture arm64:
  "_SoapySDRArgInfoList_clear", referenced from:
      _suscan_source_soapysdr_close in soapysdr.c.o
  "_SoapySDRDevice_activateStream", referenced from:
      _suscan_source_soapysdr_start in soapysdr.c.o
  "_SoapySDRDevice_closeStream", referenced from:
      _suscan_source_soapysdr_close in soapysdr.c.o
  "_SoapySDRDevice_deactivateStream", referenced from:
      _suscan_source_soapysdr_cancel in soapysdr.c.o
  "_SoapySDRDevice_enumerate", referenced from:
      _soapysdr_discovery_discovery in soapysdr.c.o
  "_SoapySDRDevice_getAntenna", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_getFrequencyRange", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_populate_device_specific in soapysdr.c.o
  "_SoapySDRDevice_getGainElement", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_populate_device_specific in soapysdr.c.o
  "_SoapySDRDevice_getGainElementRange", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_populate_device_specific in soapysdr.c.o
  "_SoapySDRDevice_getNumChannels", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
  "_SoapySDRDevice_getSampleRate", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_getSettingInfo", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_getStreamArgsInfo", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_getStreamMTU", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_hasDCOffsetMode", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_lastError", referenced from:
      _suscan_source_soapysdr_start in soapysdr.c.o
      _suscan_source_soapysdr_cancel in soapysdr.c.o
      _suscan_source_soapysdr_set_frequency in soapysdr.c.o
      _suscan_source_soapysdr_set_gain in soapysdr.c.o
      _suscan_source_soapysdr_set_antenna in soapysdr.c.o
      _suscan_source_soapysdr_set_bandwidth in soapysdr.c.o
      _suscan_source_soapysdr_set_ppm in soapysdr.c.o
      ...
  "_SoapySDRDevice_listAntennas", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_populate_device_specific in soapysdr.c.o
  "_SoapySDRDevice_listClockSources", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_listGains", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_populate_device_specific in soapysdr.c.o
  "_SoapySDRDevice_listSampleRates", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
  "_SoapySDRDevice_make", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_readStream", referenced from:
      _suscan_source_soapysdr_read in soapysdr.c.o
  "_SoapySDRDevice_setAntenna", referenced from:
      _suscan_source_soapysdr_set_antenna in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setBandwidth", referenced from:
      _suscan_source_soapysdr_set_bandwidth in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setClockSource", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setDCOffsetMode", referenced from:
      _suscan_source_soapysdr_set_dc_remove in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setFrequency", referenced from:
      _suscan_source_soapysdr_set_frequency in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setFrequencyCorrection", referenced from:
      _suscan_source_soapysdr_set_ppm in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setGainElement", referenced from:
      _suscan_source_soapysdr_set_gain in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setGainMode", referenced from:
      _suscan_source_soapysdr_set_agc in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setSampleRate", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_setupStream", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRDevice_unmake", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_close in soapysdr.c.o
  "_SoapySDRDevice_writeSetting", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
  "_SoapySDRKwargsList_clear", referenced from:
      _soapysdr_discovery_discovery in soapysdr.c.o
  "_SoapySDRKwargs_clear", referenced from:
      _suscan_source_soapysdr_close in soapysdr.c.o
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
      _strmap_to_SoapySDRKwargs in soapysdr.c.o
  "_SoapySDRKwargs_get", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
  "_SoapySDRKwargs_set", referenced from:
      _suscan_source_soapysdr_init_sdr in soapysdr.c.o
      _strmap_to_SoapySDRKwargs in soapysdr.c.o
  "_SoapySDRStrings_clear", referenced from:
      _soapysdr_discovery_populate in soapysdr.c.o
      _suscan_source_soapysdr_close in soapysdr.c.o
      _suscan_source_soapysdr_populate_device_specific in soapysdr.c.o
  "_SoapySDR_errToStr", referenced from:
      _suscan_source_soapysdr_read in soapysdr.c.o
  "_sf_close", referenced from:
      _suscan_source_config_file_is_valid in file.c.o
      _suscan_source_file_close in file.c.o
      _suscan_source_file_estimate_size in file.c.o
      _suscan_source_file_guess_metadata in file.c.o
  "_sf_command", referenced from:
      _suscan_source_config_helper_sf_format_to_str in file.c.o
  "_sf_open", referenced from:
      _suscan_source_config_sf_open in file.c.o
      _suscan_source_config_open_file_auto in file.c.o
      _suscan_source_config_open_file_raw in file.c.o
      _suscan_source_file_guess_metadata in file.c.o
  "_sf_read_float", referenced from:
      _suscan_source_file_read in file.c.o
  "_sf_seek", referenced from:
      _suscan_source_file_read in file.c.o
      _suscan_source_file_seek in file.c.o
  "_sf_strerror", referenced from:
      _suscan_source_config_sf_open in file.c.o
      _suscan_source_config_open_file_auto in file.c.o
      _suscan_source_config_open_file_raw in file.c.o
  "_sf_version_string", referenced from:
      _suscan_source_register_file in file.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libsuscan.0.3.0.dylib] Error 1
make[1]: *** [CMakeFiles/suscan.dir/all] Error 2
make: *** [all] Error 2

And I can't figure out how to fix this. I have tried uninstalling and reinstalling dependencies with arm64 architecture but am still getting the same issues.

BatchDrake, 2025-01-10:
Hi

I have been refactoring a lot of code in Suscan and this may be related to it. I am triggering a new CI build to debug it and fix it ASAP. Thanks for reporting.

In the meantime, can you confirm you are building sigutils, suscan, SuWidgets and SigDigger from `master` (and _not_ from `develop`)? We made the switch some time ago but I still have to update the docs.

IanStewart2001, 2025-01-10:
Thanks for the response. I can confirm I am building from master branch.

BatchDrake, 2025-01-11:
Hi, I've fixed a few things in SigDigger now, can you try again and tell me whether it keeps breaking? Again, everything from master (sigutils, suscan, etc)

IanStewart2001, 2025-01-11:
I rebuilt suscan from master and I got the same issue

BatchDrake, 2025-01-11:
I've tried again but I still cannot reproduce this issue. GitHub macOS machines were able to build Suscan successfully with no linker errors at all. Can you send me a copy of your build directory and all the .dylib files in your computer named "libsndfile.dylib" (the name may contain version suffixes too)? I'd like to check whether your install is right.

IanStewart2001, 2025-01-11:
I can do that sure. Im not super experienced with GutHub Issues, how would I share the whole directory with you? Also, should I rebuild all of the dependencies again from master? Reclone them all from scratch?

BatchDrake, 2025-01-11:
Nah, it's fine like that, I'd just like to look into the current state of your build directory. Zip everything and attach it to this issue. You can do it from the "Add comment" editor. You can also send it to my email address if you prefer, BatchDrake at Gmail dot com.

IanStewart2001, 2025-01-11:
[build copy.zip](https://github.com/user-attachments/files/18387467/build.copy.zip)

Here is the zipped file

BatchDrake, 2025-01-14:
Hi,

I just wanted to let you know that, after some effort, I was able to reproduce this issue in GitHub's CI machines:

```
2025-01-14T11:25:11.5805250Z ld: warning: -s is obsolete
2025-01-14T11:25:11.5805780Z ld: warning: ignoring file '/Users/runner/work/SigDigger/SigDigger/Scripts/deploy-root/usr/lib/libsigutils.0.3.0.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5806580Z ld: warning: ignoring file '/opt/homebrew/Cellar/fftw/3.3.10_1/lib/libfftw3f_threads.3.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5807400Z ld: warning: ignoring file '/Users/runner/work/SigDigger/SigDigger/Scripts/deploy-root/usr/lib/libsuscan.0.3.0.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5808420Z ld: warning: ignoring file '/opt/homebrew/Cellar/fftw/3.3.10_1/lib/libfftw3.3.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5809110Z ld: warning: ignoring file '/opt/homebrew/Cellar/libsndfile/1.2.2/lib/libsndfile.1.0.37.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5809840Z ld: warning: ignoring file '/opt/homebrew/Cellar/soapysdr/0.8.1_1/lib/libSoapySDR.0.8.1.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5810530Z ld: warning: ignoring file '/opt/homebrew/Cellar/fftw/3.3.10_1/lib/libfftw3f.3.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5811210Z ld: warning: ignoring file '/opt/homebrew/Cellar/volk/3.1.2/lib/libvolk.3.1.2.dylib': found architecture 'arm64', required architecture 'x86_64'
2025-01-14T11:25:11.5811930Z ld: warning: ignoring file '/opt/homebrew/Cellar/portaudio/19.7.0/lib/libportaudio.2.dylib': found architecture 'arm64', required architecture 'x86_64'
```

This is definitely an issue with Qt's architecture. System is arm64, but somehow SigDigger is building for x86-64. Now I know where to look for a solution, I'll keep you informed.