BatchDrake/suscan #99: Fresh Build errors out on SU_SPECTTUNER_CHANNEL_TIME_DOMAIN undeclared

jlpoolen, 2025-05-07:
I'm running Gentoo Linux on a Raspberry Pi5.  I was able to install SigDigger using Gentoo's portage packaging system, but have had an _extremely difficult time_ getting SigDigger to find my 2 RTL-SDRs connected to the Pi.  I've gone back and forth with Gentoo's ebuilds and created my own custom ebuild because I found that there were some version/path issues in the library.  My efforts to work within Gentoo's packages did not bear fruit.  It looks like the problem is at the soapy level.

So, I decided to try a from-the-wild install of SigDigger and its dependencies and followed your instructions on the [SigDigger site](https://github.com/BatchDrake/SigDigger):

       % git clone --recursive https://github.com/BatchDrake/sigutils
       % cd sigutils && mkdir -p build && cd build && cmake .. && make && sudo make install && cd ../..
       % git clone --recursive https://github.com/BatchDrake/suscan
       % cd suscan && mkdir -p build && cd build && cmake .. && make && sudo make install && cd ../..
       % git clone https://github.com/BatchDrake/SuWidgets
       % cd SuWidgets && qmake6 SuWidgetsLib.pro && make && sudo make install && cd ..
       % git clone https://github.com/BatchDrake/SigDigger
       % cd SigDigger && qmake6 SigDigger.pro && make && sudo make install && cd ..

I succeeded in building and installing **sigutils**.  I'm getting hung up on **suscan**.  I figure you have been refactoring some of your variables and maybe something has been overlooked that is causing suscan to fail.  Here's my attempt:

```
jlpoole@rpi5 /usr/local/src $ sudo rm -r suscan
jlpoole@rpi5 /usr/local/src $ git clone --recursive https://github.com/BatchDrake/suscan
Cloning into 'suscan'...
remote: Enumerating objects: 8733, done.
remote: Counting objects: 100% (1882/1882), done.
remote: Compressing objects: 100% (712/712), done.
remote: Total 8733 (delta 1284), reused 1693 (delta 1132), pack-reused 6851 (from 1)
Receiving objects: 100% (8733/8733), 5.12 MiB | 11.73 MiB/s, done.
Resolving deltas: 100% (6505/6505), done.
Submodule 'cmake/cmake-relativefilemacro' (https://github.com/antoniovazquezblanco/cmake-relativefilemacro.git) registered for path 'cmake/cmake-relativefilemacro'
Cloning into '/usr/local/src/suscan/cmake/cmake-relativefilemacro'...
remote: Enumerating objects: 15, done.        
remote: Counting objects: 100% (15/15), done.        
remote: Compressing objects: 100% (12/12), done.        
remote: Total 15 (delta 4), reused 13 (delta 3), pack-reused 0 (from 0)        
Receiving objects: 100% (15/15), done.
Resolving deltas: 100% (4/4), done.
Submodule path 'cmake/cmake-relativefilemacro': checked out '3d0e4c529dd5d83920e4299fecfb4355a4b9113c'
jlpoole@rpi5 /usr/local/src $ cd suscan && mkdir -p build && cd build && cmake ..
CMake Deprecation Warning at CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- The C compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "2.4.3")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found ZLIB: /usr/lib64/libz.so (found version "1.3.1")
-- Checking for module 'sigutils>=0.1'
--   Found sigutils, version 0.3.0
-- Checking for module 'sndfile>=1.0.2'
--   Found sndfile, version 1.2.2
-- Checking for module 'fftw3f>=3.0'
--   Found fftw3f, version 3.3.10
-- Checking for module 'SoapySDR>=0.5.0'
--   Found SoapySDR, version 0.8.1
-- Checking for module 'libxml-2.0>=2.9.0'
--   Found libxml-2.0, version 2.13.6
-- Checking for module 'volk>=1.0'
--   Found volk, version 3.1
-- Checking for module 'json-c>=0.13'
--   Found json-c, version 0.18
-- Checking for module 'alsa>=1.2'
--   Found alsa, version 1.2.13
[... 80 lines cut ...]
Date:   Sat Feb 1 13:12:45 2025 +0100

    Fix uninitialized variable in makeprof command
jlpoole@rpi5 /usr/local/src/suscan/build $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
jlpoole@rpi5 /usr/local/src/suscan/build $ 

```
I do have the STL_SDRs working and have been using them with other programs; in fact I have dedicated one for a specific frequency, 119.1 MHz, of our local airport tower, and have placed a UDP server on it then a UDP->TCP/IP server under than for programs such as openwebrx+ that cannot connect to a UDP server.  I very much would like to see SigDigger work, your "with love" quip in the credits demonstrates an ethos in a developer that should be encouraged and rewarded.  So I bring the above to your attention with the hope it will not distract you from getting your 0.4 release out the door.  

I figure that at a minimum, suscan should build, especially since it seems to with Gentoo's package:

```
jlpoole@rpi5 /usr/local/src/suscan/build $ eix suscan
[I] net-wireless/suscan
     Available versions:  (~*)0.3.0
     Installed versions:  0.3.0(15:39:15 05/05/25)
     Homepage:            https://github.com/BatchDrake/suscan
     Description:         a realtime DSP processing library

jlpoole@rpi5 /usr/local/src/suscan/build $ 
```

jlpoolen, 2025-05-07:
I confirmed tag v0.3.0 builds and see you really do have a lot of files, e.g. 151, as work in progress.  I don't expect much action on this Issue as you're still working on this, but I note this for posterity.

```
jlpoole@rpi5 /usr/local/src/suscan/build $ git branch --show-current
master
jlpoole@rpi5 /usr/local/src/suscan/build $ git tag
v0.1.0
v0.3.0
jlpoole@rpi5 /usr/local/src/suscan/build $ git checkout v0.3.0
warning: unable to rmdir 'cmake/cmake-relativefilemacro': Directory not empty
Note: switching to 'v0.3.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 37dad54 Merge branch 'develop'
jlpoole@rpi5 /usr/local/src/suscan/build $ cmake ..
CMake Deprecation Warning at CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

CppCheck found. Use cppcheck-analysis targets to run it
Cppcheck analysis already added
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /usr/local/src/suscan/build
jlpoole@rpi5 /usr/local/src/suscan/build $ make
[  0%] Building C object CMakeFiles/suscan.dir/util/cbor.c.o
[redacted to save space]
[100%] Linking C executable suscli
[100%] Built target suscli
jlpoole@rpi5 /usr/local/src/suscan/build $ 
jlpoole@rpi5 /usr/local/src/suscan/build $ git diff --stat v0.3.0..master |wc -l
151
jlpoole@rpi5 /usr/local/src/suscan/build $ 
```

BatchDrake, 2025-05-07:
Okay this is extremely odd. I will try to reproduce the build myself ASAP.